Rappetor/Sovol-SV08-Mainline #17: Question: Generic Fan 0, 1 & 3 and gcode_macro M106 & M107

Agash, 2024-07-17:
Sovol defined the part cooling fans as two separate generic fans and sets them correctly using the gcode_macro. I don't think that is necessary, as both fans aren't individually controlled, hence a virtual pin would suffice. It would also rid the sovol_macros.cfg of some lines of code and we could normally use the fans thereafter.

Hence, in the config, instead of:
```python
[fan_generic fan0] # back model cooling fan
pin: extra_mcu:PA7
max_power: 1.0

[fan_generic fan1] # front model cooling fan
pin: extra_mcu:PB1
max_power: 1.0
```
We'd have
```python
[multi_pin print_cooling_fan_pins]
pins: extra_mcu:PA7, extra_mcu:PB1

# print/part cooling fan
[fan]
pin: multi_pin:print_cooling_fan_pins
max_power: 1.0
```

Which would enable us to get rid of following lines in the sovol-macros.cfg:
```python
[gcode_macro M106]
gcode:
    {% set fan = 'fan' + (params.P|int if params.P is defined else 0)|string %}
    {% set speed = (params.S|float / 255 if params.S is defined else 1.0) %}
    {% if fan == 'fan3'%}
            SET_FAN_SPEED FAN={fan} SPEED={speed}
    {% else %}
        SET_FAN_SPEED FAN={'fan0'} SPEED={speed}
        SET_FAN_SPEED FAN={'fan1'} SPEED={speed}
    {% endif %}
    
[gcode_macro M107]
gcode:
    {% set fan = 'fan' + (params.P|int if params.P is defined else 0)|string %}
    {% if fan == 'fan3'%}
            SET_FAN_SPEED FAN={fan} SPEED=0
    {% else %}
        SET_FAN_SPEED FAN={'fan0'} SPEED=0
        SET_FAN_SPEED FAN={'fan1'} SPEED=0
    {% endif %}
```

Agash, 2024-07-17:
I was just thinking since the functionality as with these changes would be "stock" still, just more aligned with the whole mainline klipper theme, it would fit quite well. I'm eager to hear more opinions thought.

Which Discord btw, the official one or is there another?

mon5termatt, 2024-07-22:
Started working on this in #31

mon5termatt, 2024-07-22:
I haven't had the time to do the change to my printer for just using [fan] and therefore don't want to change it on the main repo yet. Once I can test that, I can make the change here.

Agash, 2024-07-22:
@Rappetor the only real place I see where another change is necessary is in the LCD config. It uses the modified M106 and M107 gcodes to set the exhaust fan speed but we can easily change that to SET_FAN_SPEED gcode, right next to the LED changes by @mon5termatt from the last PR.

mon5termatt, 2024-07-23:
Tested just [fan] last night. Seens to work just fine.

Agash, 2024-07-24:
Changed it over to [fan], fixed some minor bugs with fan0 still being referenced and added some minor QoL improvement to print_start (uncommented), print_end (uncommented) and the LCD.

Would someone be down to test this real quick? My SV08 is still in transit and probably will be for a while since it seems to move to the wrong direction lol

Agash, 2024-07-24:
then I'll wait with closing this issue until further testing even though the PR is merged.

Agash, 2024-07-24:
Neat, thank you for testing. I'll close the issue then.

closed 2024-07-24 by Agash