Rappetor/Sovol-SV08-Mainline #21: Todo: Separate File for default screen or klipperscreen

mon5termatt, 2024-07-19:
The "menu" file is not needed when an HDMI screen is connected.

Additionally some files need to be changed when running an HDMI screen. 
Currently by default the HDMI screen is upside down on this branch.

mon5termatt, 2024-07-19:
Additionally when the default screen is disconnected the beeper and screen neopixel config does not need to be in printer.cfg

mon5termatt, 2024-07-19:
Having a separate file or folder for default or HDMI could keep the printer.cfg more universal by only [include]'ing the specific config file

galosre, 2024-07-19:
I like to have both the small screen and the HDMI

Agash, 2024-07-19:
> I like to have both the small screen and the HDMI

Should be easily doable with the suggested way of OP. Generally, a more composable configuration would be beneficial for the stock experience as well as for mods.

mon5termatt, 2024-07-20:
https://github.com/Rappetor/Sovol-SV08-Mainline/pull/26

closed 2024-07-20 by mon5termatt