Rappetor/Sovol-SV08-Mainline #41: Emmc question.

sanpedrox2, 2024-08-15:


sanpedrox2, 2024-08-15:
Ok :)
im going to try them both and see what happens then :)

sanpedrox2, 2024-08-15:
I tried with that, in sd BoardEnv.txt
It didnt boot at all so changed it back.

Or do i need both fdtfile= lines?

sanpedrox2, 2024-08-15:
Ok, thanks :)
I'll try again in a while. Maybe I had something missconfigured

sanpedrox2, 2024-08-16:
Tried Some more now... Cant get it to recognize the emmc no matter what. And it doesnt boot from it either (sd card out and original OS on emmc)

Windows sees and can interact with the emmc, no problem

Suspecting my mainboard is broken maybe :/ 
I guess i'll have a chat with sovol support

Oh well, atleast it works well with the SD for the time being

sanpedrox2, 2024-08-16:
It doesnt boot with original. I have the USB adapter. I was just curious about if it could be done without. Going to mess around some more after work maybe,and see if i can get it working.

WZ9V, 2024-08-16:
I'm one of those on Discord.  The Sovol image boots fine from either the 8GB or 32GB eMMC.  The CB1 minimal or Klipper won't boot from either and cannot be seen with fdisk -l when booted from SD card.  Still have not figured it out.

sanpedrox2, 2024-08-17:
Thats strange indeed.

Shouldnt it be possible to nuke sovols klipper version from their image and install mainline on that though? And if not. Why?

sanpedrox2, 2024-08-17:
To anser my own question:
Yes and why not 🙂

You were right. The emmc works on sovols image. Flashed with rpi-imager.

I nuked sovols klipper install and the sovol user entirelly and made a new user wich user and installed kiauh. Running mainline on sovols image now and it seems to be working well

sanpedrox2, 2024-08-17:
i just did sudo userdel -rf sovol from my new user. that deletes the whole directory and the account of the user "sovol" ignoring processes the user has running.

i did also edit /etc/sudoers first so that my new user has sudo rights.

WZ9V, 2024-08-17:
I suspect my situation was a mismatch between fdtfile and the uuid.  This time I left the uuid from the original instead of copy paste.  I was about to resort to a fresh Armbian install for the CB1.

Rotule666, 2024-08-20:
I tried fresh Armbian and the emmc problem is still there, when trying to boot, I have the emmc data error on the picture.
[image]

also notice the PCB version
[image]

The stock image was booting allrigth, I overwritten it on the emmc because I tought it was a problem with my emmc V1.0 I tried to use but it does the same with eMMC V1.1 from the SV08.

Fallback is to use the SD card, but.

sanpedrox2, 2024-08-20:
Try stock image. And install mainline klipper on that if all else fails. That worked for me.
I made a new user but im not sure that is needed. But I would uninstall everything klipper related using kiauh on stock. Then delete kiauh folder and then follow this guide from step 4 - install mainline klipper

sanpedrox2, 2024-08-20:
Btw. I got best result using raspberry pi imager (no device selected, use custom os. No saved configuration) 

That seems to be what sovol recommends aswell

Rotule666, 2024-08-20:
I would try this, anyone have a dump of the stock image? lol...

sanpedrox2, 2024-08-20:
Sovol has it on their page

cyaneo, 2024-08-27:
I want use a 2nd eMMC module but can't find a eMMC V1.2 (where to buy?) - but I have a eMMC v1.0 32g laying arround - could I use this also?

ss1gohan13, 2024-08-28:
> I want use a 2nd eMMC module but can't find a eMMC V1.2 (where to buy?) - but I have a eMMC v1.0 32g laying arround - could I use this also?

Yes, this should work.

sanpedrox2, 2024-08-28:
Nice 😀
I have sort of moved on now. Got it setup on a 32g no issues whatsoever. Amd my curiosity has dimmed a bit. Now I just want to print 😉

Good for future tinkerers though 👍

RazorSiM, 2024-12-03:
I know it has been closed, but I can confirm the new `dtb` works correctly :) 

**This applies to anyone booting from SD card that now wants to move to eMMC.**

TLDR: follow the [step 3](https://github.com/Rappetor/Sovol-SV08-Mainline?tab=readme-ov-file#step-3---changes-to-the-boardenvtxt--setup-wi-fi) again :) 

If you followed the guide before the latest updates, it was mentioned to use a `biqu dtb` and place it inside the `/boot/dtb/allwinner` folder. After you connect the eMMC to the board and run `fdisk -l`, you'll notice your device is not there at all. What you have to do is to get [the new dtb](https://github.com/Rappetor/Sovol-SV08-Mainline/tree/main/files-used/dtb-files), put it in your `boot/dtb/allwinner` folder, change the `BoardEnv.txt` accordingly and you're good to go.