Rappetor/Sovol-SV08-Mainline #47: Checking details before publishing video

teachingtechYT, 2024-08-30:
Hi there,
I'm converting to mainline Klipper using this excellent guide as part of my series on converting the SV08 to be a budget toolchanger:
https://www.youtube.com/playlist?list=PLGqRUdq5ULsNIN_40Bv1iTbElj6O7BRrU
I plan to cover this guide in an upcoming video. I've opted to keep the original eMMC as a backup install, and set up everything onto an SD card instead of eMMC.

So far I'm up to updating MCU firmware and things have been pretty smooth. There were a few things where I had small issues, so I just wanted to check I hadn't done anything wrong.

1. I inverted the HDMI output in system.cfg as specified. During boot, the on screen console in upside down, but KlipperScreem is correct. Is that normal?

2. I had to install gcode shell command through Kiauh to keep some of the config files I restored happy.

3. Other files I restored had references to files being in home/sovol/.. rather than the new home/biqu/.. structure. I manually found these and updated them and Klipper is happy now.

In part 2 of my series, I did some customisation of the .cfg files, so it's possible I have made my life harder by trying to restore these files rather than stick exactly to the guide.

Thanks in advance for any guidance.

mon5termatt, 2024-08-30:
Yeah the upside down boot screen is normal.

teachingtechYT, 2024-08-31:
Thanks for the comprehensive reply.

I think my issues with restoring files will be fixed when the final guide is written and others follow it in a better order. That should make it straightforward for others.

I'm having to use the ST Link alternative software with my clone programmer. After some fluffing about I can now connect and read, although this older version of the software doesn't have 'read all'. Can you possibly help with me the exact values for the memory range for getting the exact 128kb of flash saved as a backup?

[image]

I'm close with trial and error but nor quite there.

teachingtechYT, 2024-08-31:
I think I'm there with
[image]

THat gives a file 128kb (or 131,072 bytes) in size.

ss1gohan13, 2024-09-01:
> Yeah the upside down boot screen is normal.

I can work on that I think

ss1gohan13, 2024-09-01:
> Hi there, I'm converting to mainline Klipper using this excellent guide as part of my series on converting the SV08 to be a budget toolchanger: [...]

I'm currently at a wedding. But we welcome you too mainline. 

@Rappetor  any concern about another video? Or is there going to be a collaboration?

teachingtechYT, 2024-09-02:
Everything is now completed and connected, although I haven't run a print to verify.

I think I have a repeatable workflow available for the older ST link software to show in the video. It was finicky until I found the right settings.

More feedback:

1. Does it make sense to install, build and retrieve the katapult bootloader binary BEFORE turning off the printer to do the STM programmer work? If only to avoid turning it off to backup, back on to prepare katapult, then back off to flash katapult.

2. I had to modify `update_klipper_mcus_sv08.sh` for multiple toolheads. This worked but my changes were quite inefficient. To accommodate people with one toolhead or multiples like me, could there be a toolhead variable at the top as well as a place to input multiple serial ids? Then during the toolhead phase, the flashing loops through the specified number of toolheads doing one at a time? Perhaps a sanity check to see that if X nuber of toolheads are desired, that there are X ids added, ready to go. My current method is slow because you need to build identical toolhead MCU firmare multiple times.
```
#!/usr/bin/env bash

#Replace each serial number with the one you find in your printer.cfg file
#HOSTSERIAL = [mcu]
#TOOLHEAD0SERIAL = [tool0]
#TOOLHEAD1SERIAL = [tool1]
#TOOLHEAD2SERIAL = [tool2]
#TOOLHEAD3SERIAL = [tool3]
#TOOLHEAD4SERIAL = [tool4]
HOSTSERIAL='30FFD4053347543034871651-if00'
TOOLHEAD0SERIAL='51FF6D064849824936420767-if00'
TOOLHEAD1SERIAL='51FF6A064849824934540167-if00'
TOOLHEAD2SERIAL='51FF70064849824929290167-if00'
TOOLHEAD3SERIAL='51FF73064849824936430767-if00'
TOOLHEAD4SERIAL='51FF72064849824931160167-if00'

#COLORS
MAGENTA=$'\e[35m\n'
YELLOW=$'\e[33m\n'
RED=$'\e[31m\n'
CYAN=$'\e[36m\n'
NC=$'\e[0m\n'
NC0=$'\e'

#SCRIPT COMMAND DEFINITION
cd "$HOME/klipper"

stop_klipper(){
	echo -e "${YELLOW}Stopping Klipper service.${NC}"
	sudo service klipper stop
}

start_klipper(){
	echo -e "${YELLOW}Starting Klipper service.${NC}"
	sudo service klipper start
}

flash_host(){
	cd "$HOME/klipper"
	echo -e "${YELLOW}Step 1/4: Cleaning and building Klipper firmware for Host MCU.${NC}"
	make clean KCONFIG_CONFIG=host.mcu
	read -p "${CYAN}Check on the following screen that the parameters are correct for the ${RED}HOST${NC0}${CYAN}firmware. Press [Enter] to continue, or [Ctrl+C] to abort.${NC}"
	make menuconfig KCONFIG_CONFIG=host.mcu
	make KCONFIG_CONFIG=host.mcu -j4
	mv ~/klipper/out/klipper.bin host_mcu_klipper.bin
	read -p "${CYAN}Host MCU firmware building complete. Please check above for any errors. Press [Enter] to continue, or [Ctrl+C] to abort.${NC}"
	echo -e "${YELLOW}Step 2/4: Flashing Klipper to Host MCU.${NC}"
	cd ~/klipper/scripts/ && python3 -c 'import flash_usb as u; u.enter_bootloader("/dev/serial/by-id/usb-Klipper_stm32f103xe_'$HOSTSERIAL'")'
	sleep 3
	~/katapult/scripts/flashtool.py -f ~/klipper/host_mcu_klipper.bin -d /dev/serial/by-id/usb-katapult_stm32f103xe_$HOSTSERIAL
}

flash_toolhead0(){
	cd "$HOME/klipper"
	echo -e "${YELLOW}Step 3/4: Cleaning and building Klipper firmware for Toolhead 0 MCU.${NC}"
	make clean KCONFIG_CONFIG=toolhead.mcu
	read -p "${CYAN}Check on the following screen that the parameters are correct for the ${RED}TOOLHEAD${NC0}${CYAN}firmware. Press [Enter] to continue, or [Ctrl+C] to abort.${NC}"
	make menuconfig KCONFIG_CONFIG=toolhead.mcu
	make KCONFIG_CONFIG=toolhead.mcu -j4
	mv ~/klipper/out/klipper.bin toolhead_mcu_klipper.bin
[... 65 lines cut ...]
}

#SCRIPT EXECUTION
stop_klipper

flash_host

flash_toolhead0

flash_toolhead1

flash_toolhead2

flash_toolhead3

flash_toolhead4

read -p "${CYAN}Toolhead MCU firmwares flashed, please check above for any errors. Press [Enter] to continue, or [Ctrl+C] to abort.${NC}"

start_klipper

cd "$HOME/klipper"
```

There's quite a few steps, but each of them is simple and well documented. Well done to everyone involved for this great guide and resource.

teachingtechYT, 2024-09-03:
That looks ideal. Thanks for your time.

The video should be coming out Saturday 14/9 morningAus time.

K2AAE, 2024-09-04:
I have a box full of programers, 6 years ago I purchased a jlink and stlink and haven't spent anymore time solving issues my cheapness caused. 
 STLINK-V3PWR $95 USB C Has everything plus Power management to nano amp measurements 
 STLINK-V3SET $35 USB Micro B Need B-STLINK-ISOL board for 1.65 Voltage and Isolation $40 or STLINK-VOLT 1.65 voltage $21.71
 STLINK-V3MINIE $11.70 USB C
NOTE: All the above parts use STLINK Version 3 and work with STM32CubeProgrammer without issues 
mouser.com

mon5termatt, 2024-09-06:
> I have a box full of programers, 6 years ago I purchased a jlink and stlink and haven't spent anymore time solving issues my cheapness caused. STLINK-V3PWR $95 USB C Has everything plus Power [...]

I mean, that's a lot more money than a cheap ST link from Amazon.