Rappetor/Sovol-SV08-Mainline #50: run_current and hold_current issue

aquari86, 2024-09-07:
your printer.cfg doesn't work well with my sv08, the printer always lost z-offset, even while it was printing it got worse layer by layer, I think it was the print.cfg, run_current and hold_current were changed there, which caused the printer to lose height, I changed all the values ​​of run_current and hold_current to the original values ​​from sovol and it works again. Thank you for your great work and the effort you put in.

sziryboy, 2024-09-07:
> a printer.cfg-d nem működik jól az sv08-ammal, a nyomtató mindig veszített z magasságból, még nyomtatás közben is rétegről rétegre romlott, szerintem a print.cfg volt, ott megváltozott a run_current és a hold_current, ami miatt hogy a nyomtató elveszítse a magasságát, megváltoztattam a run_current és hold_current összes értékét a sovol eredeti értékére, és újra működik. Köszönjük nagyszerű munkáját és fáradozását.

thanks! You have saved me from a nervous breakdown. I followed your instructions and my printer is fine.

sziryboy, 2024-09-07:
> I guess we can up the current a bit on the motors. their rated (max) current is 0.8A. So 0.8 * 0.707 = 0.56. So let's make it 0.55. I have changed the default config. Hopefully it will help some people in the future as well.
> 
> Another option could be to disable stealthchop (change to stealthchop_threshold: 0 or comment it out). That could also help in fixing skipping steps.
> 
> I would also check for binding or other issues in the Z. Because 0.45 should be quite enough to move the Z (like 4 of the motors going at it). What worked a bit with me was to loosen everything up (the flying gantry, all the screws for the Z pillars, the top frame) and re-tighten in a different order than what the manual says: first the flying gantry, then the top frame and last the Z pillar screws.
> 
> Then again, upping the run_current from 0.45 to 0.55 is much more easy 😄

Why is hold_current in the comment?
Thank you for your excellent work!

sziryboy, 2024-09-07:
So now I understand 😄
Thank you very much!