Rappetor/Sovol-SV08-Mainline #51: Latest klipper (28995a8bc) seems to break sv08

vvuk, 2024-09-13:
Just moved to mainline klipper, and with very latest klipper (28995a8bc) I am getting a ton of "timer too close" errors trying to do anything. (Immediately triggered just trying to soak the bed; triggered partway through z offset calibration, etc.)

Looking at commits that landed today, they look suspicious -- some new gcode queueing functionality. I went back to `293858` which looked to be just before this and the issues went away.

nick119, 2024-09-13:
Thanks for posting this, i too am having the same problem. Downgraded to 293858 as well, no issues.

mon5termatt, 2024-09-13:
Suggest making a issue on Klipper as well.

vvuk, 2024-09-13:
Can't; they don't do issues, I'd have to jump into their discord and I didn't really have time to engage :|

vvuk, 2024-09-14:
This seems to be fixed in very latest sovol (they fixed the regression;  we just happened to get lucky and installed on the broken day!)

closed 2024-09-14 by vvuk