Rappetor/Sovol-SV08-Mainline #60: Pressure probe and z offest

Yousafkhan1976, 2024-09-21:
Thank you for a perfect setup guide, I was able to get to the end without errors.

Now when I try to run z offset calibration, I get an error when the SAVE_CONFIG command executes.

"SAVE_CONFIG section 'probe' option 'z_offset' conflicts with included value"

ss1gohan13, 2024-09-21:
Please link your config file. The stock sovol config should not result in this issue.

Yousafkhan1976, 2024-09-21:
I have all the includes in my printer.cfg

[include mainsail.cfg]
[include sovol-macros.cfg]
[include timelapse.cfg]
[include options/lcd/*.cfg]
[include options/probe/stock.cfg]

Yousafkhan1976, 2024-09-22:
I moved the [probe] section to printer.cfg and commented out the z-offset line. Now it works

closed 2024-09-22 by Yousafkhan1976