Rappetor/Sovol-SV08-Mainline #61: update_klipper_mcus_sv08.sh not showing all necessary options

BoaTheCrafter, 2024-09-21:
When running the modified update_klipper_mcus_sv08.sh, the menuconfig screen doesn`t show all options mentioned in step 8.8 and the corresponding note. 
Notable missing options are the bootloader offset and the communication interface as seen here:

[image]

If you still run the script, it ends with the Error 

`mv: cannot stat '/home/biqu/klipper/out/klipper.bin': No such file or directory`

The Windows file formatting issue can be rolled out, since i used a Arch Linux based notebook for configuration.

The only changes, i've made to the script are as follows:

[image]

The script is placed and executed in the ~/klipper directory

Yousafkhan1976, 2024-09-21:
You need to change the Microcontroller Architecture to STM32

BoaTheCrafter, 2024-09-21:
Oh my god. How did i overlook that lol

BoaTheCrafter, 2024-09-21:
Currently not at home, but i'd guess so. Gonna close this, once i tried it

BoaTheCrafter, 2024-09-22:
Yeah, fixed it. Guess i had just some brain lag

closed 2024-09-22 by BoaTheCrafter