Rappetor/Sovol-SV08-Mainline #62: Installing BDSensor

Yousafkhan1976, 2024-09-25:
I am attempting to install the BDSensor on top of my mainline Klipper install. https://github.com/markniu/Bed_Distance_sensor/wiki/Installing-for-Klipper

If I add the required BDSensor section in my printer.cfg, the system doesn't boot. First error was "...[probe_pressure] section must have z-offset..]. I changed the BDSensor section to probe then rebooted and uncommented the [probe_pressure] section.

Now the error is "Option 'pin' in section 'probe' must be specified", but the guide for BDSensor doesn't have any setting for "pin". The pins are defined as Sda_pin and scl_pin which are set to PB6 and PB7 respectively.

Yousafkhan1976, 2024-09-25:
I have up on the idea for now and ran the auto configure script again to flash the mcus back to mainline Klipper. I'm having issues now with the probe not triggering. The light on the stock probe works (goes off when near metal), but query_probe output is "open" even when when it should be triggered (probe light off).

I reverted config backup from the last known good printer.cfg and stock.cfg, but nozzle still crashes into the bed.

Yousafkhan1976, 2024-09-27:
I fried the extra_mcu board trying to figure out the issue. Now waiting for a new one. When the replacement is installed I will check if the issue persists and report back.

Yousafkhan1976, 2024-10-04:
I have installed the new toolhead and the printer is functional again. I will attempt the BDSensor I stall again next week and update.

theminor, 2025-03-10:
> I have installed the new toolhead and the printer is functional again. I will attempt the BDSensor I stall again next week and update.

@Yousafkhan1976 Were you able to resolve this issue? I just got a BDSensor and now I'm hesitant to install it in the SV08 if it won't work and/or will fry my toolhead board!