Rappetor/Sovol-SV08-Mainline #68: Question: enclosure exhaust fan handling

real-alphamaennchen, 2024-09-29:
What is the best way to get the exhaust fan of the enclosure to work as it worked  in stock sovol version?
I thought of creating a macro to turn it on and add that to the start code of the filament types, where it is needed.
Is there a better way to get it to work?

Big thanks for all contributors for this great repo!

mak0t0san, 2024-09-30:
There is a line (75) in sovol_macros.cfg that can be uncommented out to turn on the exhaust fan during print start.
https://github.com/Rappetor/Sovol-SV08-Mainline/blob/13103e6fac65462a0227c21759d46c0c7c77fcac/files-used/config/sovol-macros.cfg#L75

real-alphamaennchen, 2024-10-02:
but this would enable it always, not filament type dependent.
I thought of the setting in OrcaSlicer for the exhaust fan, i guess it uses the "M106 P3 S255" command. But the "P3" also activates the part cooling fans.
Is there a better way to enable that feature than switch to the sovol solution of the stock firmware?

ss1gohan13, 2024-10-03:
You need to do a bit of modifications to the start print macro, install a [thermistor](https://github.com/ss1gohan13/Sovol-SV08-Mainline/blob/main/Extras/Chamber%20Thermistor/BMP280_Readme.md), and then call the fan to activate within the start print macro.

I refer you to my macros (start print is in there): [Macro](https://github.com/ss1gohan13/sv08/blob/main/printer_data/config/macro.cfg)

I've adopted it from [here](https://github.com/jontek2/A-better-print_start-macro)

Val-dose-things, 2024-11-12:
i had to switch back to fan0 and fan 3. the Gcode was not starting the exhaust fan. it sends M106 p3 S90 but P3 is defined in the M106 macros.  the following works with no issues. 

[multi_pin print_cooling_fan_pins]
pins: extra_mcu:PA7, extra_mcu:PB1

[fan_generic fan0] # print/part cooling fan
pin: multi_pin:print_cooling_fan_pins
max_power: 1.0

[fan_generic fan3] # exhaust fan
pin: PA2
max_power: 1.0

[gcode_macro M106]
gcode:
    {% set fan = 'fan' + (params.P|int if params.P is defined else 0)|string %}
    {% set speed = (params.S|float / 255 if params.S is defined else 1.0) %}
    {% if fan == 'fan3'%}
            SET_FAN_SPEED FAN={fan} SPEED={speed}
    {% else %}
        SET_FAN_SPEED FAN={'fan0'} SPEED={speed}
    {% endif %}
    
[gcode_macro M107]
gcode:
    {% set fan = 'fan' + (params.P|int if params.P is defined else 0)|string %}
    {% if fan == 'fan3'%}
            SET_FAN_SPEED FAN={fan} SPEED=0
    {% else %}
        SET_FAN_SPEED FAN={'fan0'} SPEED=0
    {% endif %}