Rappetor/Sovol-SV08-Mainline #70: Control MCUs with external computer

alliefitter, 2024-10-10:
Hi, I'm waiting on parts to come in before I mainline my SV08, and I've been reading the instructions ahead of time. I was wondering if it's possible to just flash the main MCU and the toolhead and then control them with a Klipper instance running on an external computer? If it is possible, would I be losing things like the camera, z offset sensor, etc..? Thanks in advance.

Cowboy124aa, 2024-10-10:
No you dont lose functionallity of the camera or z offset sensor, this mainline klipper streamlines the system, and in one case makes the stepper motors run cooler, The whole purpose of the Klipper is to run you printers from one or more computer, pretty much anywhere that has internet access, similar to octoprint everywhere but I personally think Klipper/mainsail is much better.  Also if you plan on adding more then one tool head you do have to have mainline Klipper to run multiple ones because Sovol's factory klipper doesnt support additional ones.

froh42, 2024-10-10:
@alliefitter You probably can't run the MCU from an external computer, as MCU and the ARM SBC are on a single board and (I think) the serial port between both is not available externally. But even if you don't have the emmc yet, you can run a fresh install from an SD card (the SBC side of the mainboard has an SD card slot and can boot from it rather from the emmc).

In fact I have the emmc here now, but haven't got around to install it as my SV08 runs Klipper mainline from the SD just fine.

closed 2024-10-10 by alliefitter