Rappetor/Sovol-SV08-Mainline #79: Input shaping Error

senseilangford, 2024-10-23:
I got every thing working as far as i can tell but the input shaping give me the following issues.
On the hdmi screen the autocalibration is there but it seem none functional.
Then I get  ACCELEROMETER_QUERY and invalid command!
Also is the automatic z off set that sovol used available now? As i don't see it

geekko, 2024-10-23:
Did you flash the toolhead and the mainboard with your klipper firmware as described in step 8 and did you also "uncomment the [adxl345] and [resonance_tester] parts in your printer.cfg" as described in point 12 of step 8?

As far as I understand the printer should work without flashing the firmware but the accelerometer won't work, that's why it's commented out by default.

senseilangford, 2024-10-23:
You were 100% right I didn’t miss that. What about the Z offset compared to Sobel default and after mainline is there a way to have it do it just like it did on the civil default??Sent from my iPhone