Rappetor/Sovol-SV08-Mainline #103: Printing above bed after BTT Eddy calibration

WayneManion, 2024-11-27:
I installed Klipper mainline. I had some issues with getting the z-offset set properly. I wanted faster bed meshes for this big printer anyway, so I ordered a BTT Eddy USB. I printed Rappetor's Eddy bracket model out of some nice teal ASA and mounted it. I believe I followed the instruction properly. 

Some steps I took:
- removed the [probe] section from printer.cfg
- moved the comment indicators around to disable probe and enable eddy
- edited the eddy.cfg file with my probe's serial identifer

The probe is recognized. I can use set_kinematic_position to position the Eddy (the "horizontal" PCB) about 20mm above the print bed. I run `LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy` and get a value. I accept the value, save, restart Klipper. I have done this a few times, it pretty much always comes back as 15. 

I heat up the nozzle and retract the filament. I wait a little while and then clean the nozzle with a little copper brush. I run `PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy`. I use the wizard to get the nozzle to the point where a piece of paper is almost trapped by the nozzle. I have done this several times and the printer always ends up printing higher than it should, so I err slightly on the side of 'too tight' for the paper test. 

At this point, the printer is always printing between 0.5 and 1.0mm above the correct z-height for a good first layer. I'm printing test patches with PETG and a first layer height of 0.36mm on a 0.6mm CHT nozzle in a Micro Swiss FlowTech hot end made as a drop-in replacement for SV08. PETG should stick really well to the textured PEI. The 0.36 layer height should be very forgiving, but even still the test patches have great big gaps between the lines. When I babystep the nozzle down by ~0.75mm things start looking great. 

[image]

I've tried adding and subtracting the amount I babystep the printer into the `eddy.cfg [probe_eddy_current btt eddy] z_offset` value, but that doesn't seem to fix the problem. 

Some wild guesses:

The "horizontal" PCB of the Eddy isn't actually all that parallel to the bed. I don't know much about how these probes work, but maybe it is important that it is dead parallel to the bed?

I have a Micro Swiss FlowTech nozzle with a 0.6mm CHT nozzle installed. My measurements show the FlowTech hot end is within a couple tenths of a millimeter in length (z) compared to the stock hot end. The nozzle outlet is in exactly the same position as the stock hot end with respect to x & y. I don't think this makes a difference, but who knows. I know that my test patch gcode is sliced for a 0.6 nozzle, and a first layer height of 0.36mm, which should be extra forgiving.

butterbutts, 2024-12-01:
I was just having the same problem. No matter how I set up the current calibration it ends up too high. I ended up importing the beta z-offset cfg from the eddy repo, changing the include to use that instead, and manually copied all of the config values from the built in eddy.cfg to the beta config.

WayneManion, 2024-12-03:
This is the config that ended up working for me (beta z-offset)

`# The MCU section only applies to the Eddy USB. For Eddy Coil you will use the MCU name of the toolboard that you connected the Eddy Coil to.
[mcu eddy]
serial: /dev/serial/by-id/usb-Klipper_rp2040_4550357128914B98-if00  # Change this to your own path
restart_method: command

[temperature_sensor btt_eddy_mcu]
sensor_type: temperature_mcu # Sets the type of sensor for Klipper to read
sensor_mcu: eddy # Sets the MCU of the eddy probe tempereature sensor
min_temp: 10 # Sets the minimum tempereature for eddys tempereature sensor to operate
max_temp: 100 # Sets the maximum tempereature for eddys tempereature sensor to operate

[probe_eddy_current btt_eddy]
sensor_type: ldc1612
#i2c_address:
i2c_mcu: eddy  # This value is good for the Eddy USB but would need to be adjusted for the Eddy Coil according to the MCU you have used.
i2c_bus: i2c0f # This value is good for the Eddy USB but would need to be adjusted for the Eddy Coil according to the I2C bus you have used.
# Measure the offsets below using the method described here: https://www.klipper3d.org/Probe_Calibrate.html#calibrating-probe-x-and-y-offsets
# For a standard Voron stealthburner X carriage mount there should be no need to change the defaults below.
x_offset: -16.5
y_offset: 11.5
z_offset: 2

# This section is only relevant for Eddy USB. Comment it out for Eddy Coil.
[temperature_probe btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26
horizontal_move_z: 2

[bed_mesh]
horizontal_move_z: 2
speed: 200
mesh_min: 15, 15
mesh_max: 330, 330
probe_count: 9, 9
algorithm: bicubic

# Uncomment this if you are using Eddy as the probe AND the homing endstop
[safe_z_home]
home_xy_position: 172, 172
z_hop: 10
z_hop_speed: 25
speed: 200

###############################Macros and related################################
#This secton contains macros and related config sections. Some should be used, others are optional. Read the comment above each one to find out whether or not to uncomment it for your installation.

# Uncomment this if you are using Eddy as the probe AND the homing endstop AND would like to use the beta z-offset control
[save_variables]
filename: ~/printer_data/config/variables.cfg

# Uncomment this if you are using Eddy as the probe AND the homing endstop
[force_move]
enable_force_move: True # Allows a user to move the z axis down if they have no other means of homing Z and need to calibrate the Eddy.

# Uncomment this if you are using Eddy as the probe AND the homing endstop AND would like to use the beta z-offset control
[delayed_gcode RESTORE_PROBE_OFFSET]
initial_duration: 1.
gcode:
  {% set svv = printer.save_variables.variables %}
  {% if not printer["gcode_macro SET_GCODE_OFFSET"].restored %}
    SET_GCODE_VARIABLE MACRO=SET_GCODE_OFFSET VARIABLE=runtime_offset VALUE={ svv.nvm_offset|default(0) }
    SET_GCODE_VARIABLE MACRO=SET_GCODE_OFFSET VARIABLE=restored VALUE=True
  {% endif %}

# Uncomment this if you are using Eddy as the probe AND the homing endstop
# Take note of the lines that should only be used if you have a KNOMI installed.
[gcode_macro G28]
rename_existing: G28.1
[... 40 lines cut ...]
    {% endfor %}
    {% set rawparams=paramList|join(' ') %}
    SET_GCODE_VARIABLE MACRO=SET_GCODE_OFFSET VARIABLE=runtime_offset VALUE=0
  {% endif %}
  SET_GCODE_OFFSET_ORIG { rawparams }

# This macro automates a lot of the frequency mapping process and simplifies the steps significantly.
[gcode_macro PROBE_EDDY_CURRENT_CALIBRATE_AUTO]
gcode:
  BED_MESH_CLEAR
  G28 X Y
  G90 # Abs positioning
  G1 X{ printer.toolhead.axis_maximum.x/2 } Y{ printer.toolhead.axis_maximum.y/2 } F6000
  {% if 'z' not in printer.toolhead.homed_axes %}
    SET_KINEMATIC_POSITION Z={ printer.toolhead.axis_maximum.z-1 } # Allows the user to work it down until it touches.
  {% endif %}
  PROBE_EDDY_CURRENT_CALIBRATE {rawparams}

#This macro is optional but useful if you want to run a rapid scan before each print. Simply uncomment it and add BED_MESH_SCAN to your print start code.
[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BTT_BED_MESH_CALIBRATE
gcode:
#  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=True #Only uncomment this line if using a KNOMI and then remove the BED_MESH_CALIBRATE macro from KNOMI.cfg
  BTT_BED_MESH_CALIBRATE METHOD=rapid_scan
#  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False #Only uncomment this line if using a KNOMI and then remove the BED_MESH_CALIBRATE macro from KNOMI.cfg`

onefast50, 2024-12-09:
Thanks for all this .....I been fighting that issue for days. My other 4 printer are using the Th3d inductive probe and are flawless and though I'd give the Eddy and try. Going to try the z_offset beta for a last ditch effort to get it working.