Rappetor/Sovol-SV08-Mainline #105: Flashing Main MCU - Stuck on attempting to connect to bootloader when Mainsail has a connection.

Wrath669, 2024-12-01:
Is there's a web session for mainsail or any app running that's talking to Klipper, it seems that Klipper doesn't close cleanly as this prevents the main MCU flashing. At stage 8.8, he command line just hangs on "Attempting to connect to bootloader".
Rebooting doesn't solve, if the connected devices connect again after the SV08 board reboots.
This does not affect flashing the toolhead MCU, that flashes without fault.
2 users have found this fault, so it is consistently repeatable.
https://www.reddit.com/r/SovolSV08/comments/1h3hgxg/comment/lzqq8np/?context=3

A notice in the instruction to close any web interfaces on connected devices would be beneficial.

Another user was able to get an error instead of the CLI session just hanging:
ERROR:root:Error on serial read Traceback (most recent call last): File "/home/biqu/katapult/scripts/flashtool.py", line 540, in _handle_response data = self.serial.read(4096) File "/home/biqu/.local/lib/python3.9/site-packages/serial/serialposix.py", line 595, in read raise SerialException( serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?) ERROR:root:Flash Error Traceback (most recent call last): File "/home/biqu/katapult/scripts/flashtool.py", line 624, in run await flasher.connect_btl() File "/home/biqu/katapult/scripts/flashtool.py", line 99, in connect_btl ret = await self.send_command('CONNECT') File "/home/biqu/katapult/scripts/flashtool.py", line 216, in send_command raise FlashError("Error sending command [%s] to Device" % (cmdname)) FlashError: Error sending command [CONNECT] to Device During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/home/biqu/katapult/scripts/flashtool.py", line 678, in main await sock.run(args.device, args.baud, fpath) File "/home/biqu/katapult/scripts/flashtool.py", line 631, in run await flasher.finish() File "/home/biqu/katapult/scripts/flashtool.py", line 295, in finish await self.send_command("COMPLETE") File "/home/biqu/katapult/scripts/flashtool.py", line 216, in send_command raise FlashError("Error sending command [%s] to Device" % (cmdname)) FlashError: Error sending command [COMPLETE] to Device

chrisf4lc0n, 2024-12-05:
Hi,
After connecting the ST-Link, have you fully erased the MCUs, built the katapult and flashed it into the MCUs without any errors?

DjDecent, 2024-12-05:
Yes

nitromaroder, 2024-12-09:
Same issue here, toolhead flash was successful, mcu cannot be flashed (full erase, same katapult.bin as for the toolhead):

> Bus 006 Device 002: ID 1d50:6177 OpenMoko, Inc. stm32f103xe
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               2.00
>   bDeviceClass            2 Communications
>   bDeviceSubClass         0
>   bDeviceProtocol         0
>   bMaxPacketSize0        16
>   idVendor           0x1d50 OpenMoko, Inc.
>   idProduct          0x6177
>   bcdDevice            1.00
>   iManufacturer           1 katapult
>   iProduct                2 stm32f103xe
>   iSerial                 3 32FFDB054747363029790257
>   bNumConfigurations      1

Also verified that "ttyACM1" ("/dev/serial/by-id/usb-katapult_stm32f103xe_32FFDB054747363029790257-if00 -> ../../ttyACM1") is not in use ("lsof|grep ttyACM1"), which is in use after executing the flash by the Python script. Not sure, what the issue could be...

DjDecent, 2024-12-09:
The answer is in the issue. Make sure you close any browser/app connections that might be open with the machine and try again. :)

nitromaroder, 2024-12-09:
> The answer is in the issue. Make sure you close any browser/app connections that might be open with the machine and try again. :)

There is nothing open, and as I mentioned before, prior to flashing the mcu, "lsof" doesn't show any connection to the ttyACM1, so the issue is somewhere else

nitromaroder, 2024-12-09:
Ok, I got the issue sorted @Wrath669 - I have updated yesterday my ST Link already, but obviously there was another update - after updating to the latest version and a new flash of "katapult.bin", the mcu flash was successful (I also compared the output and this time the chip contained little bit more of data). 

Hope this helps!

[image]

DjDecent, 2024-12-09:
> > The answer is in the issue. Make sure you close any browser/app connections that might be open with the machine and try again. :)
> 
> 
> 
> There is nothing open, and as I mentioned before, prior to flashing the mcu, "lsof" doesn't show any connection to the ttyACM1, so the issue is somewhere else

So that means it's not the same issue we were describing here. It would be better to start a new issue. If you include the versions and files you were using I will try to duplicate the error so we can be sure it was the update that was causing the problem.

nitromaroder, 2024-12-09:
@DjDecent - you're right, I'll try to find out which version I had before and will open a new issue. Thanks for the swift response!

doesntfazer, 2025-01-23:
I know this issue is closed, but I would like to note the version firmware on the stlink that makes it fail. 
It's version V2J46S7.
You should be on version V2J45S7 and update it from the firmware upgrade button within STM32CubeProgrammer.
@Rappetor This was a major issue, and I think it should for sure be mentioned in the documentation.

foxshared, 2025-12-09:
@doesntfazer tq bro u fix my problem change the firmware to V2J46S7 make the installation pass....btw sv06 ace toolhead firmware update....