Rappetor/Sovol-SV08-Mainline #106: Can not update MCU 'extra_mcu' config as it is shutdown

MrD003, 2024-12-03:
Hey there,
I just finished the ReadMe.md and did all the steps as described.

when i try to calibrate z-offset i am getting this error on the mainsail dashboard:

[image]

[moonraker (2).log](https://github.com/user-attachments/files/17999112/moonraker.2.log)
[klippy (2).log](https://github.com/user-attachments/files/17999113/klippy.2.log)

does anyone know why i am getting this error and how to fix it?

Best Regards

chrisf4lc0n, 2024-12-05:
This looks like you have not flashed the MCUs properly or serial IDs for MCUs are incorrect.

MrD003, 2024-12-05:
i think i solved it, as one fan connector was only partly connected.

closed 2024-12-05 by MrD003