Rappetor/Sovol-SV08-Mainline #107: Printer not booting or ssh

MrD003, 2024-12-05:
I have ran the installation process once and wanted to do it again.
So i went ahead and flashed the original bootloaders on the mcu and toolhead. I then reflashed the sovol image onto the emmc.

So far so good. 
The printer booted up normally.

I then went ahead and reinstalled the btt image on the emmc, changed the BoardEnv.txt, added the dtb file and put my wifi info into the settings.cfg.

After this nothing seemed to work anymore. The ethernet connection doesn't show any traffic, the device doesnt seem to boot at all and ist stuck in this position. The lcd says booting but even if the mcu was alive it couldn't interact with the lcd yet so that is not an indicator.

Can someone help me?

chrisf4lc0n, 2024-12-05:
Are your serial numbers for the MCUs correct in the printer.cfg? Also since you cannot SSH to the printer have you followed the instructions for setting the WLAN up?

MrD003, 2024-12-05:
Hey there, 
The printer.cfg is not yet there, i only have the BOOT directory in which i added the .dtb to dtb/allwinner/ . 
I also added the lines to the BoardEnv.txt files which i pulled from a backup of a working sovol installation.
The printer is connected to my network via ethernet.

chrisf4lc0n, 2024-12-05:
Do not add any lines from the original Sovol BoardEnv.txt.
Please follow the STEP 3 to the "T".
Copy sun50i-h616-sovol-emmc.dtb to /dtb/allwinner/.
REPLACE the content of the BoardEnv.txt above #-----# with:
```
bootlogo=false
overlay_prefix=sun50i-h616
fdtfile=sun50i-h616-sovol-emmc
console=display
overlays=uart3
overlays=ws2812
overlays=spidev1_1
```

MrD003, 2024-12-05:
yes i have done this.

bootlogo=false
overlay_prefix=sun50i-h616
fdtfile=sun50i-h616-sovol-emmc
console=display
overlays=uart3
overlays=ws2812
overlays=spidev1_1
#------------------------------------------------#
rootdev=UUID=b6d595f7-d1b2-4e72-ba8f-3d3efe7bb147
rootfstype=ext4

[image]

But you are supposed to take the BoardEnv.txt from the Backup and then modify that one right?

chrisf4lc0n, 2024-12-05:
> But you are supposed to take the BoardEnv.txt from the Backup and then modify that one right?
>
I am afraid not, you only make a copy of the BoardEnv.txt just in case and replace the content entirely with what is in STEP 3.
Do you mind me asking how did you pull the sun50i-h616-sovol-emmc.dtb?

MrD003, 2024-12-05:
directly from the link out of step 3 from the repo: 
"Please copy the sun50i-h616-sovol-emmc.dtb file from the /files-used/dtb-files/ GitHub folder [HERE](https://github.com/Rappetor/Sovol-SV08-Mainline/blob/main/files-used/dtb-files) to your 'BOOT' drive /dtb/allwinner/ folder."

Then i will flash it again and use the BoardEnv.txt from the image

chrisf4lc0n, 2024-12-05:
Did you download the sun50i-h616-sovol-emmc.dtb RAW?
>Then i will flash it again and use the BoardEnv.txt from the image
>
Yes, please.

MrD003, 2024-12-05:
nope i downloaded it normally.

will update you

MrD003, 2024-12-05:
nope still not working sadly. I used the original .txt version from the image and downloaded the the dtb raw...

stil nothing

MrD003, 2024-12-05:
if i load the sovol image it seems to be working and booting fine, which is a bit weird

MrD003, 2024-12-05:
Okay good news, i flashed everything and i mean EVERYTHING again, MCU, Toolhead, completely formated the EMMC with two different tools, just to make sure (redundancy and stuff), deleted every downloaded file from the repo, redownloaded everything, then reflashed and added/changed what had to be and now IT WORKS :') 

Thank you for all the inputs! 
i am not sure what it was im just glad it seems to be working now :)) 

BTW @Rappetor really really good repo! besides this incident even a i could follow it which says a lot ^^