Rappetor/Sovol-SV08-Mainline #130: Missing options/hdmi

gregma, 2025-01-04:
I finally got my btt hdmi5 and installed klipperscreen.  I changed the include to [include options/hdmi/*.cfg]
 and restarted.  screen did nothing.  Then I realized that there was no options/hdmi folder.  Should I be getting that from somewhere else?

JarTheHead, 2025-01-05:
If you moved to mainline recently, it should work without options/hdmi just by plugging in the screen and installing KlipperScreen. This was recently changed in #99 to account for users that use both an HDMI screen and the stock LCD.

Is the screen backlight turning on?

gregma, 2025-01-05:
Right now it shows a screen:

Don't forget to check you added the save_variables section in your printer.cfg! You NEED to do that for things to work correctly!

With a "LETS GET STARTED!" button in the lower right. Clicking that does nothing.

Nothing else, and nothing is clickable.  This never changes.  Kiauh shows klipperScreen  v0.4.5-39 installed.  I do have save_variables in printer.cfg.  Everthing else looks good.  Only other weirdness is that the LED scroll wheel is now backwards.  used to be scroll right and it went down, now scroll right moves the cursor up.

gregma, 2025-01-05:
Well, after another power cycle, it seems to now be at the menu.  However, nothing is clickable.

closed 2025-01-05 by gregma