Rappetor/Sovol-SV08-Mainline #135: Part Fan at 60% After Prints

zimmra, 2025-01-10:
Things have been working great but I can't seem to figure out why my part fan is going to 60% after each print. 

I initially thought it was because I set PA6 to be initialized on startup when flashing the firmware to the toolhead board, so I re-flashed it (excluding PA6 form startup gpio) and it still behaves this way.

My end G-Code in Orca is just set to `END_PRINT`

The complete `END_PRINT` sequence below runs

```
[gcode_macro END_PRINT]
description: 
variable_state: 'normal'
gcode:
    {% set z_max = printer['gcode_macro _global_var'].z_maximum_lifting_distance|int %}
    {% set e_mintemp  = printer.configfile.settings['extruder'].min_extrude_temp %}

    M400
    SET_GCODE_VARIABLE MACRO=START_PRINT VARIABLE=state VALUE='"Prepare"'
    SET_GCODE_VARIABLE MACRO=START_PRINT VARIABLE=record_extruder_temp VALUE=0  
    SET_GCODE_VARIABLE MACRO=START_PRINT VARIABLE=max_record_extruder_temp VALUE=0

    M117 Done
    G91
    {% if printer['filament_switch_sensor filament_sensor'].enable == True and
          printer['filament_switch_sensor filament_sensor'].filament_detected == True
    %}
        {% if (printer.extruder.target != 0 and printer.extruder.temperature >= printer.extruder.target) or
              printer.extruder.temperature >= e_mintemp
        %}
            G1 E-2 F2700
            G1 E-2 Z0.2 F2400
        {% endif %}
    {% endif %}

    {% if (printer.gcode_move.position.z + 25) < z_max %}
        G1 Z+25 F3000
    {% else %}
        G1 Z+{(z_max - printer.gcode_move.position.z)} F3000
    {% endif %}
    G90
    M117 Parking
    G1 X0 Y360 F9000

    TURN_OFF_HEATERS
    M117 Fans, Heaters and Motors Off
    _ALL_FAN_OFF
    M84 X Y Z E  

    M220 S100
    M221 S100

    CLEAR_PAUSE
    M117 Complete!

    {action_respond_info("Finish Print!")}

[gcode_macro _ALL_FAN_OFF]
gcode:
    M106 S0
    # Run the exhaust fan for another 2min after the other fans and then turn it off
    # UPDATE_DELAYED_GCODE ID=exhaust_fan_off DURATION=120
    M107
```

Running both `M106 S0` and `M107` both shut the fan off when ran manually. 

Any clue what's going on here?

JarTheHead, 2025-01-10:
Try double checking your slicer's End Gcode

JarTheHead, 2025-01-11:
Ah yes, [Orca will use M106 to try to control the exhaust fan](https://github.com/SoftFever/OrcaSlicer/wiki/air-filtration). Since the fans and M106 weren't configured with this in mind, it just sends the fan speed to your part cooling fan. Unchecking the box should work.

zimmra, 2025-01-12:
This was the issue - thanks @JarTheHead and @Rappetor. Thought I looked through all the settings, always forget there's some more hidden within then filament profile too.

closed 2025-01-12 by zimmra

AnyOldName3, 2025-09-14:
I don't think this should have been closed. It's reasonable to expect that if you follow the instructions in this repo, you'll end up with an SV08 that worked like it did before given that the guide has sentences like *Do a firmware_restart (or reboot the whole printer) and you should have a working SV08.* As Sovol provide a definition of M106 that makes the extractor fan work, I would have expected this repo's `sovol-macros.cfg` to have one, too.