Rappetor/Sovol-SV08-Mainline #137: no IP after emmc flash

Jaro3D, 2025-01-15:
hello I followed the steps from method 1 but at very begin I meet problem and don't know what I am doing wrong. 
let me tell what I did step by step 
I purchased emmc v2 adaptor from amazon MKS Makerbase
I turned printer off and set it on its back
I took the cover off and then remove 2 screws holding chip.
gently pulled out chip wearing antistatic gloves and fit it in to the reade/adaptor
(for additional information) I tried with balena etcher and with rasbery pi imager 
I flashed emms with CB1_Debian11_minimal_kernel5.16_20240319.img.xz
I replaced text in BoardEnv.txt 
didn't set up wifi as I have it connected via ethernet cable 
I copied file sun50i-h616-sovol-emmc.dtb to allwiner folder and safely remove storage from windows
after this I installed bac emmc to the printer and turned it on. 
CB1_Debian11_Klipper_kernel5.16_20240319 .img
here is where I meet problem cannot see printer in my network. 
I tried with sd but same result not sure what to do next.
will be greatfull for any advices

mon5termatt, 2025-01-16:
Have you ran an IP scan? It's the same mac so in theory it should use the same IP

FunkyOwl-Net, 2025-01-18:
I faced exactly the same problem and could fix.

I copied the system.cfg to my desktop, renamed it to an .txt file and wrote in my Wifi settings without the ".
Then i renamed it back to a .cfg file and copied to the EMMC

Jaro3D, 2025-01-23:
> Did you fix this issue already?
>
> I'm not 100% sure where it might go wrong. First make sure it's actually
> booting, if you have an hdmi display or something that can be used you
> might get more useful information that way. That's always the difficult
> part here, there is no display to check what is going on unless you plug in
> some hdmi display.
>
> Otherwise you should be able to find the device on your router. As far as
> I know ethernet should work by default, but you can also try wifi just to
> be sure. Although as far as I know both should work (ethernet and wifi).
>
> So hope you got it figured out already, if so would gladly hear what the
> solution was.

I have managed to load stock to emmc printer works back again. There was
problem with klipper and moonraker and I couldn't reinstall them with
success. Someone on one of the groups shared link to Sovol stock emmc image

ulloveit, 2025-02-03:
I'm having same problem tried 2 different emmc cards and no luck. I have a hdmi screen and nothing comes up on that either.

ulloveit, 2025-02-04:
Figured it out... If you are coping and pasting from the instructions it has a bad line in it "fdtfile=sun50i-h616-sovol-emmc" the problem is the "sovol" part it is referencing files that aren't there. It needs to be "fdtfile=sun50i-h616-biqu-emmc" that fixed my issue. Hope it helps.

Bug-president, 2025-04-23:
This was also my experience, and I tried the network info as a string, a single quote string, and a literal on both the stock emmc and an SD card. I do feel that it's worth mentioning that the home directory and all the Klipper files weren't present on the emmc when I went to flash it the first time. In fact, the preloaded image looked a lot like the image given in the instructions. Attempting to boot with a screen attached yielded no results. Restoring to the stock image worked just fine. 

While ssh'd into the printer I can indeed see all the klipper files. Excuse my lack of knowledge - what's keeping us from switching the repo on the preinstalled KIAUH and continuing on from there?

Bug-president, 2025-04-23:
>>Mind you, it will always show 'booting' because that is just an image (shown until Klipper initializes the screen, which is not going to happen initially because there is no Klipper installed yet with the 2.3.4 minimal CB1 image). 

Understood! I should have been clearer, I am referring to an external HDMI screen. I'm not getting any indication that it's booting. I have the BTT screen arriving later today, I'll give it another try at that point as my main display setup has a few complications between the source and the output that may have been getting in the way. 

>> Just make sure to use the 2.3.4 minimal CB1 image, copy over the sun50i-h616-sovol-emmc.dtb after flashing, change the BoardEnv.txt (and make sure to KEEP the red part, already in your BoardEnv.txt, as shown in the image and replace the green part). Enter your wifi credentials in the system.cfg and it should work.

That's the method I've been following - direct from the instructions in your repo. Nothing seems to be happening unfortunately, but on the stock image identical network credentials work just fine! 

>>I do feel that it's worth mentioning that the home directory and all the Klipper files weren't present on the emmc when I went to flash it the first time. In fact, the preloaded image looked a lot like the image given in the instructions.

What do you make of this? Something to be concerned about? Or is it just partitioned in a strange way? When I plug the emmc into my computer I can see the whole storage volume in the one drive but the klipper files seem to be elsewhere in my case. I'll do more looking through the filesystem later on today via ssh. 

I think my most likely solution at this point (if I can't get the new image to work on the emmc or the SD card) would be to flash the stock image on a 256gb SD card to give myself some storage headroom and updating through switching the repo on KIAUH - then proceeding through the steps from the point where mainline klipper is installed.

Bug-president, 2025-04-24:
I actually think I accidentally figured it out - I was downloading the dtb file by right clicking and saving as. When I tried to do that with the python scripts I found that it prepended a bunch of HTML data which caused a syntax error in klipper. Since I'm so far along I don't think I'm going to test it, but I think that may be it.

mikehatch, 2025-11-02:
I'd just had this issue as well with the sun50i-h616-sovol-emmc.dtb file being corrupted - I simply clicked the link in the instructions to download it, but it was ~200k as covert says.  

I downloaded the repo and copied the file and it was ~40k as indicated, and the sovol came up on the network finally.  

You should probably update the instructions so that the 'easiest path' (clicking on the link provided) doesn't go down a dead end that is very hard to diagnose.  I spent 2 hours reflashing before coming here to see this.  

Thanks for this though!  I appreciate the work that goes into providing a guide like this.