Rappetor/Sovol-SV08-Mainline #146: Poor Z Homing but excellent probe repeatability with Eddy

klack, 2025-01-26:
When I home Z, the height comes back with a variation of about +-0.06.

I have achieved a probe repeatability test of +-0.01 so this is confusing
I have done Drive Current, and Thermal Drift Calibrations for Eddy
This problem seems to exist no matter the temperature of the bed or toolhead

Steps to reproduce:

1. Home Z
2. Observe Z Height
3. Z Height will show normal, sometimes +- 0.03 and sometimes up to +-0.06

Sample output:
```
Homing Z
probe at 177.500,177.500 is z=2.596014
Result is z=2.596014
Homing Z
probe at 177.500,177.500 is z=2.560919
Result is z=2.560919
<snip>
Result is z=2.561552
Result is z=2.579780
```
```
PROBE_ACCURACY
PROBE_ACCURACY at X:177.500 Y:177.500 Z:7.460 (samples=10 retract=2.000 speed=5.0 lift_speed=5.0)
probe at 177.500,177.500 is z=2.574636
probe at 177.500,177.500 is z=2.579179
probe at 177.500,177.500 is z=2.580705
probe at 177.500,177.500 is z=2.579979
probe at 177.500,177.500 is z=2.580340
probe at 177.500,177.500 is z=2.578026
probe at 177.500,177.500 is z=2.578113
probe at 177.500,177.500 is z=2.578115
probe at 177.500,177.500 is z=2.578696
probe at 177.500,177.500 is z=2.576767
probe accuracy results: maximum 2.580705, minimum 2.574636, range 0.006069, average 2.578456, median 2.578406, standard deviation 0.001714
```

Other things I've tried:
- Increasing Probe Samples to 5
- Lowering Samples Tolerance
- Running PROBE command manually
- Lowering `z_offset` in eddy config
- Testing completly cold, and completly hot
- Turning off bed mesh
- Meds

Can anyone reproduce this?  Does anyone know why I might have poor Z homing, but good probe repeatability?

klack, 2025-01-26:
Closing this as it is klipper related:
https://hackmd.io/yEF4CEntSHiFTj230CdD0Q?view

klack, 2025-01-27:
Reopened because my issue is not effected by thermals

klack, 2025-01-29:
Thank you for the suggestions.  I tried slowing down the homing speed, and non-beta config, as well as removing all the homing scripts and going with safe_homing.  Unfortunatly I received the exact same results.  I will be trying out an M12 probe in the time being.

klack, 2025-01-31:
I hooked up the M12 probe and simply switched from stock.cfg to eddy.cfg.  Problem went away :(

vvuk, 2025-01-31:
Z homing with eddy sensors (any eddy sensor) is not intended to be super precise.

The way homing works is the printer keeps slowly moving the toolhead down, checking endstop switch state after every tiny move. With eddy sensors, there is no physical switch, so the way it works is the eddy sensor is told "hey notify the host when the frequency goes above this value [frequency is mapped directly to height]".

The toolhead slowly moves down, the eddy sensor is constantly checking frequencies, and at some point it goes above the trigger frequency. It then sends a command back to the klipper host saying "hey I triggered back at this time".. and only then does klipper stop the move. There's some fudging then that happens to figure out where the toolhead was at that time and call that the home position. But it's expected that it won't be exactly the trigger height.

If you want a "more accurate" home, you can do something like (old code):
```
G28 Z
PROBE
SET_Z_FROM_PROBE
```
(note: BTT's eddy code [sort of does this](https://github.com/bigtreetech/Eddy/blob/d5665df6576097b3a514d326070ee6b9859eb51e/sample-bigtreetech-eddy-homing.cfg#L87) but does it in a not useful/good way -- it just sets the z-offset from the probe vs actually homing the axis)

or (with eddy-ng):
```
G28 Z
PROBE_EDDY_NG_PROBE_STATIC HOME_Z=1
```

the home_z=1 bits say "ok now that Z is roughly homed, do a probe with the eddy, and then set Z to be exactly the height it says the toolhead is at right now".

vvuk, 2025-01-31:
Also even if you use BTT's macros as-is, the value returned by `PROBE_ACCURACY` will still be different -- again, because all they set is the gcode offset, vs. actually homing the axis.

klack, 2025-01-31:
@vvuk Thanks for the suggestions, but I have manually done in the past:

PROBE
G1 Z5
PROBE
G1 Z15
PROBE
G1 Z15

And it returns the exact same +-0.06 inaccuracy.

The eddy can find Z 100% of the time in the probe accuracty test (within 0.15 as expected).
There should no inaccuracy when doing the the same move manually.

klack, 2025-02-07:
Move to https://github.com/bigtreetech/Eddy/issues/120

closed 2025-02-07 by klack