Rappetor/Sovol-SV08-Mainline #158: Step 8 Issue

Npott93, 2025-02-26:
When I reach step 8 and then make it to step 8 in the sub section, I'm having issues with klipper trying to flash to the host mcu. I go through the programming step just fine but when it comes to flash, it just sits at "Flashing Serial Device /dev/serial/by-id/usb-katapult_stm32f103xe_35FFD6054748303416732457-if00, baud 250000
Detected USB device running Katapult" and doersn't do anything. I have tried doing everything over and I am still coming up witht the same issue.

[image]

ning-y, 2025-02-27:
I ran into this same issue with Mainsail open (because I was coping the serial IDs). Closing the Mainsail tab and rebooting the printer fixed it for me.

Npott93, 2025-02-27:
I am so upset right now because I did have the mainsail tab open when trying to flash klipper. After it wouldn’t work I tried flashing the the host mcu and tool head mcu back to stock along with a fresh emmc install. Now my board won’t connect to the internet via wireless or hard wired. Any idea on how to fix it?

ning-y, 2025-02-27:
Sorry to hear that... I decided to mainline Klipper on my SV08 because after I ran `apt upgrade` on the stock eMMC I had the same problem you've described, in which the mainboard OS fails to connect to network (the ethernet port on my router indicated the SV08 was not connected). My issue was fixed by following this guide, so it's possible that you can fix yours by restarting from step 1. I admit, it's a pain, though. Maybe someone else can advice a better solution.

Npott93, 2025-02-28:
I don't have an hdmi screen, just the stock one so it's locked at the svol booting screen. Im trying to reflash the emmc now.

Npott93, 2025-02-28:
Oh my lord. It honestly helps to follow the directions as they are written. Did not add the file into the allwinner folder. Going to flash everything and I'll update when I get to the spot I got stuck at.

Npott93, 2025-02-28:
Success! Printer is working now! I guess having mainsail open was the issue.