Rappetor/Sovol-SV08-Mainline #167: flashing klipper to host hangs, last step.

gkouros50, 2025-03-01:
When i attempt to flash kipper to the host mcu the process just hangs for what seems like far too long.

[image]

here are my MCUs and make menu configs

[image]

ACM1 is the host 
ACM0 is the tool head

as you can see i've successfully been able to get kipper flashed to the tool head

gkouros50, 2025-03-01:
should i uninstall klipper for the flashing then?

gkouros50, 2025-03-01:
I've re-flashed katapult and now am getting Attempting to connect to boot-loader 

[image]

olie-o, 2025-03-02:
I'm having the same issue. In my case I was able to flash the host, but get stuck on "Detected USB device running Katapult" when flashing the toolhead.

olie-o, 2025-03-02:
> I'm having the same issue. In my case I was able to flash the host, but get stuck on "Detected USB device running Katapult" when flashing the toolhead.

Well, I just did one last try before going to bed and got it working. I went back to step 7.2 "cd ~/katapult && make menuconfig" and proceeded from there. 

The only thing I did differently I think was selecting "full chip erase" from the left hand side menu in STM32 programmer, not from the button on the right hand side in the erase menu. Not sure if that was it, but I'm up and printing again. 

[image]

gkouros50, 2025-03-03:
I can confirm i went back ensured to erase chip and reflashed katuplut-> then klipper and all went well. Please REMOVE YOUR ST-LINK BEFORE ATTEMPTING TO FLASH. nothing bad should happen but that can also cause it to hang on 'detected'

gkouros50, 2025-03-03:
New to Github issues apologizes for reopening, was attempting to mark @olie-o as the solver to the issue

closed 2025-03-03 by gkouros50