Rappetor/Sovol-SV08-Mainline #168: Some commands on stock screen aren't working.

Npott93, 2025-03-02:
I tried using the screen to use some of the commands and they won't do anything. I also tried useing the auto calibrate feature and after the input shaper ran it threw the shut down prompt. Is there something in the config file that I missed?

[image]

Npott93, 2025-03-04:
I do have the adxl and resonance tester ucommented. Auto calibrate keeps throwing that mcu shut down, Belt tester does the same thing.

HuskyRiver, 2025-03-14:
I have the same issue. Auto Calibrate made it through x but crashed after starting y after 2Hz like above.  The [adxl345] and [resonance_tester] sections are uncommented. I ran SHAPER_CALIBRATE on the console and it completed fine. Not sure what's different with menu choice.