Rappetor/Sovol-SV08-Mainline #172: Numpy installation wont work & ls -la /dev/serial/by-id/ not working either

CoinNester, 2025-03-09:
Hello, 
I have been trying to mainline my SV08 for days now and my problems first started with ls -la /dev/serial/by-id/ not working at all. 

lsusb does work but it does not show the toolhead being connected because when I disconnect it, it doesn't change anything:

pi@mainsailos:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pi@mainsailos:~ $ ls -la /dev/serial/by-id/
ls: cannot access '/dev/serial/by-id/': No such file or directory

So I kept trying to reinstall Debian using balenaEtcher and also Raspberry Pi imager and nothing has worked. Followed all the directions.

Now I also have a problem with:

~/klippy-env/bin/pip install -v numpy

I get the following error even though I haven't done anything out of the ordinary:

  error: subprocess-exited-with-error

  × Building wheel for numpy (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /home/pi/klippy-env/bin/python3 /home/pi/klippy-env/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py build_wheel /tmp/tmp8dc65_bc
  cwd: /tmp/pip-install-ztopj6i8/numpy_c311b1d047a84842b0cf68cd94b44f48
  Building wheel for numpy (pyproject.toml) ... error
  ERROR: Failed building wheel for numpy
Failed to build numpy
ERROR: Could not build wheels for numpy, which is required to install pyproject.toml-based projects

Could somebody please help me with both of these problems? Thank you in advance.

Val-dose-things, 2025-03-14:
https://www.thushanfernando.com/posts/2023/missing-serial-by-id/

SuddenImpact58, 2025-03-21:
I tried this "fix" & it does not work. I have 4 decades of PC exp including building from scratch.

Val-dose-things, 2025-03-21:
last person i helped fix this he was editing the name of 
CB1_Debian11_minimal_kernel5.16_20240319.img.xz 
to 
CB1_Debian11_minimal_kernel5.16_20240319.img
make sure you are not editing this file name.

SuddenImpact58, 2025-03-22:
That would be me Val.... Ask a pro.... lol

SuddenImpact58, 2025-03-22:
But it still does not work.

CoinNester, 2025-04-15:
Does anyone have any ideas about this problem? It seems that some people have it and some don't while doing the same exact install. Makes no sense to me. The fix above does not work as stated by some one else.

Val-dose-things, 2025-04-20:
The problem seems to be with the creation of the klipper bin. using a pre existing one seems to bypasses the issue.

Val-dose-things, 2025-04-20:
i cant upload the working .bin i have so i don't know where they got theirs.

Val-dose-things, 2025-04-20:
looks like they figured it out
https://github.com/Rappetor/Sovol-SV08-Mainline/issues/178#issuecomment-2800150081