Rappetor/Sovol-SV08-Mainline #173: Klipper reports: ERROR - No section: 'Variables'

N3rdyBlacksmith, 2025-03-14:
Hi folks.  Just reaching out to see if other folks have had this issue.  I've attempted to go through docs a few times, but I keep getting this same error.  I've gone through the config files that I can think to check, but I'm not seeing anything that is trying to call "Variables".  I've seen the saved_variables.cfg, but no luck.  Any help would be greatly appreciated.  I changed the extensions to .txt so that I can post.  Thanks,

-Steve

[image]

[klippy.log](https://github.com/user-attachments/files/19253959/klippy.log)

[printer.txt](https://github.com/user-attachments/files/19253961/printer.txt)

[saved_variables.txt](https://github.com/user-attachments/files/19253962/saved_variables.txt)

[sovol-macros.txt](https://github.com/user-attachments/files/19253963/sovol-macros.txt)

[KlipperScreen.txt](https://github.com/user-attachments/files/19253965/KlipperScreen.txt)

[moonraker.txt](https://github.com/user-attachments/files/19253967/moonraker.txt)

aclundwall, 2025-03-29:
I am also getting this error.  I updated my printer to Mainline Klipper yesterday, and things seemed to be going fine.  Through several reboots everything was OK.  Then I attempted a Z-offset calibration, and the printer failed during the pressure probing.  After that, I'm getting the same error you're seeing.

aclundwall, 2025-03-29:
Hey, thanks for replying!  
I found my problem too, although I'm still not sure exactly what was going on.  Like I said, I installed Mainline Klipper yesterday.  Once it was up and running, I added an eddy USB to the mix (or tried to).  Things were going OK, but I think I damaged the Eddy USB when I took it out of it's case (to mount it to the adapter).  It was found and flashed OK when I installed it, but I suspect I broke the fragile solder joints at the base because I bent it getting it out of the plastic housing.  That seemed to be confirmed when I tried to do the drive current calibration, and it came back with zero volts.  So I put the stock probe back in, put my config files back to the way they were, and went back to business.  I went through a couple restarts seemingly without issue before I attempted to do the Z offset calibration.  Klipper shut down upon the failure of that calibration, and subsequent restarts were when I saw the error here.  The logs indicated it couldn't find a device, but the only 'devices' I was expecting were the MCU and Toolhead MCU, and they were both working.  But it turns out that when I removed the Eddy from the toolhead, I neglected to actually unplug the USB cable, so it was still plugged in.  Once I unplugged that, the error went away (and the Z-offset completed successfully).  I'm going to revisit this in a few days when I get a replacement Eddy USB.