Rappetor/Sovol-SV08-Mainline #177: Automatic MCU update script will not work

SuddenImpact58, 2025-03-17:
Follow every step & still get  the same error. . Preformed the trouble shooting  made sure no dos format. Still get....Connecting to Serial Device /dev/serial/by-id/usb-katapult_stm32f103xe_38FFD405334E4B3124640751-if00, baud 250000
ERROR:root:Flash Tool Error
Traceback (most recent call last):
  File "/home/biqu/katapult/scripts/flashtool.py", line 1096, in main
    await sock.run()
  File "/home/biqu/katapult/scripts/flashtool.py", line 1033, in run
    await self.validate_device(device)
  File "/home/biqu/katapult/scripts/flashtool.py", line 925, in validate_device
    raise FlashError(f"No Serial Device found at {dev_path}")
FlashError: No Serial Device found at /dev/serial/by-id/usb-katapult_stm32f103xe_38FFD405334E4B3124640751-if00

[config-20250317-165242.zip](https://github.com/user-attachments/files/19302306/config-20250317-165242.zip)

[klippy.log](https://github.com/user-attachments/files/19302288/klippy.log)

Trying to get this done for almost a week. 

Thank you

thrystan, 2025-04-23:
I am having a similar error, when I ls /dev there is no /dev/serial available. It was there before I flashed the mcus.

thrystan, 2025-04-24:
> I am having a similar error, when I ls /dev there is no /dev/serial available. It was there before I flashed the mcus.

Never mind, I'm an idiot. I build the katapult.bin for the wrong device, I used the defaults instead of selecting the stm micro-controller.