Rappetor/Sovol-SV08-Mainline #178: Klipper reports: Error

NightWasher, 2025-04-06:
After going through the steps up to step 5. Everything is installed, but I get an error message (mcu 'mcu' : Unable to connect.) I have gone back through and redid all the steps three times now. And every time it ends with this. I don't know what could have been wrong or what to do to fix it. 

[image]

NightWasher, 2025-04-08:
Sorry for not including this in the first post. Hope this helps. 

[klippy (1).log](https://github.com/user-attachments/files/19646824/klippy.1.log)

[moonraker.log](https://github.com/user-attachments/files/19646868/moonraker.log)

NightWasher512, 2025-04-08:
When I SSH into the printer and put in this command ls /dev/serial/by-id/ 
It's says no location found.

NightWasher, 2025-04-10:
I haven't had time to really work on this sense the last comunication. But these are the screen shots of my SSH. I looks like I have the deb11u1 and not the u2. 

[image]

[image]

[image]

NightWasher512, 2025-04-10:
I'm just wanting to get a functional printer back. Do you think that a new emmc would fix it. Or a new main board.

NightWasher512, 2025-04-10:
Do you think a new emmc would fix it, or would I have to buy a new Sovol main board?

NightWasher, 2025-04-12:
I did these steps that you said "Another thing to try just to rule some stuff out, if you kinda have easy access to the emmc, is to re-flash the CB1 image (in short re-do Steps 1,2 & 3) but do not run sudo apt update && sudo apt upgrade. And the first thing you do, also with the hdmi 7 disconnected, once you get SSH access for the first time is to try ls /dev/serial/by-id/ and see if it's (still) there at this point." and I still got the same thing. I even then got a Mico SD card and flashed that. And I did the boot up just form the card with the emmc not installed. And I got the same thing. And just to be 100% sure. I followed your steps word for word (even though I have now done this so many times I have them memorized) and I get the same thing. 

[image]

NightWasher, 2025-04-12:
I did try going through and installing Kiauh. And then going to advanced menu and then #5 [get MCU ID] and then 1) USB and it said it could not see anything. I then tried 2) UART and it came back with this. 

[image]

NightWasher, 2025-04-13:
So after looking at discord and deciding to flash the MCU and the Tool head. And then I ls /dev/serial/by-id/ and they showed up like it should with the katapult instead of Klipper. Then after finishing the process it all works. One thing I will add to anyone that ever has this problem and is reading this, make sure you look at the log window in the STM32CubeProgrammer. Because when I did the download on the MCU the first time it did not say "download complete". But when I clicked on the Toolhead MCU and downloaded it. I then saw that it said "download complete". So I scrolled back through the log and saw that there was no other placed where it said that. So I hooked back up to the MCU and did it again, and that time it said "download complete". I don't know if that will help anyone, but just food for thought. 

[image]

closed 2025-04-13 by NightWasher