Rappetor/Sovol-SV08-Mainline #183: KIAUH not instaling klipper

chaz2000, 2025-04-26:
Hi, i have got up to the point of using KIAUH to install Klipper. However, it is not completing the install. I have tryed starting over a fue times now to make sure I'm not missing a step but i get the same error red text each time i try to install klipper:

DEPRECATION: Building 'python-can' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'python-can'. Discussion can be found at https://github.com/pypa/pip/issues/6334
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.9/py_compile.py", line 144, in compile
    code = loader.source_to_code(source_bytes, dfile or file,
  File "<frozen importlib._bootstrap_external>", line 853, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/aenum/_py2.py", line 5
    raise exc, None, tb
             ^
SyntaxError: invalid syntax

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/compileall.py", line 238, in compile_file
    ok = py_compile.compile(fullname, cfile, dfile, True,
  File "/usr/lib/python3.9/py_compile.py", line 150, in compile
    raise py_exc
py_compile.PyCompileError:   File "/home/biqu/klippy-env/lib/python3.9/site-packages/aenum/_py2.py", line 5
    raise exc, None, tb
             ^
SyntaxError: invalid syntax

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 105, in _run_wrapper
    status = _inner_run()
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 96, in _inner_run
    return self.run(options, args)
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/req_command.py", line 68, in wrapper
    return func(self, options, args)
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/commands/install.py", line 459, in run
    installed = install_given_reqs(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/req/__init__.py", line 83, in install_given_reqs
    requirement.install(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/req/req_install.py", line 867, in install
    install_wheel(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/operations/install/wheel.py", line 728, in install_wheel
    _install_wheel(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/operations/install/wheel.py", line 614, in _install_wheel
    success = compileall.compile_file(path, force=True, quiet=True)
  File "/usr/lib/python3.9/compileall.py", line 255, in compile_file
    msg = err.msg.encode(sys.stdout.encoding,
TypeError: encode() argument 'encoding' must be str, not None

[ERROR] Error installing Python requirements: Installing Python requirements failed!
[ERROR] Error during installation of Klipper requirements!
[ERROR] Error installing Python requirements: Installing Python requirements failed!
[ERROR] Klipper installation failed!

It is also worth noting I fumbled my way through to step 6 the first time I tried, a week ago. But i had problems with the plugins, so I decided to start over befor flashing the KATAPULT BOOTLOADER.

chaz2000, 2025-04-26:
Thanks for the first response. 
I did the updates and get the KIAUH GIT as described in walkthrough. I believe it's the new KIAUH because it asks. I will look at using the olded version.
 I have re-flashed the emmc lots of times now so once more isn't an issue :)

chaz2000, 2025-04-26:
I was using KIAUH 6.0.0 beta.2. I just successfully installed everything using the older version v5.1.4 (by choosing option 2 on KIAUH startup). Once I had installed everything, I went back and used the new v6 to install Numpy.  
Thank you for the support, I really appreciate all the work done here!

closed 2025-04-26 by chaz2000

s3anami, 2025-04-27:
I am having the same issue and going back to the stable version doesn't fix anything

chaz2000, 2025-04-28:
FYI, the issue was not KIAUH as I initially thought, and has now been fixed with the workaround dated 2025-04-28. I now have everything working using KIAUH v6