Rappetor/Sovol-SV08-Mainline #184: Error with Python install with Kiauh

s3anami, 2025-04-27:
Traceback (most recent call last):
  File "/usr/lib/python3.9/py_compile.py", line 144, in compile
    code = loader.source_to_code(source_bytes, dfile or file,
  File "<frozen importlib._bootstrap_external>", line 853, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/aenum/_py2.py", line 5
    raise exc, None, tb
             ^
SyntaxError: invalid syntax

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/compileall.py", line 238, in compile_file
    ok = py_compile.compile(fullname, cfile, dfile, True,
  File "/usr/lib/python3.9/py_compile.py", line 150, in compile
    raise py_exc
py_compile.PyCompileError:   File "/home/biqu/klippy-env/lib/python3.9/site-packages/aenum/_py2.py", line 5
    raise exc, None, tb
             ^
SyntaxError: invalid syntax

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 105, in _run_wrapper
    status = _inner_run()
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 96, in _inner_run
    return self.run(options, args)
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/cli/req_command.py", line 68, in wrapper
    return func(self, options, args)
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/commands/install.py", line 459, in run
    installed = install_given_reqs(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/req/__init__.py", line 83, in install_given_reqs
    requirement.install(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/req/req_install.py", line 867, in install
    install_wheel(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/operations/install/wheel.py", line 728, in install_wheel
    _install_wheel(
  File "/home/biqu/klippy-env/lib/python3.9/site-packages/pip/_internal/operations/install/wheel.py", line 614, in _install_wheel
    success = compileall.compile_file(path, force=True, quiet=True)
  File "/usr/lib/python3.9/compileall.py", line 255, in compile_file
    msg = err.msg.encode(sys.stdout.encoding,
TypeError: encode() argument 'encoding' must be str, not None

s3anami, 2025-04-27:
dependency version mismatch per kiauh maintainer, wasn't the nicest about answering but here is what was said
https://github.com/dw-0/kiauh/issues/676

s3anami, 2025-04-27:
what are the discords mentioned in the tutorial and here by the way?

Cogsworth03, 2025-04-28:
Nevermind! I followed the same troubleshooting steps used for the klipper installation, and it seems to have taken care of the problem. Thank you so much. It just occurred to me that I used your guy's guide for flashing my FIRST 3D printer, the Davinci 1.0. Wow, how far things have come. Thanks again!

Cogsworth03, 2025-04-28:
Any idea what this is about?

[image]

v-tron, 2025-04-28:
> Any idea what this is about?
> 
> [image]

I just pressed connect again and it worked.

v-tron, 2025-04-28:
> Ok, I was able to re-create this issue. And it's indeed a depedency issue with Klipper, I guess they need to fix it. Workaround: as per above instructions (from kiauh), edit the `klippy-requirements.txt` and change `python-can==3.3.4` to `python-can~=4.3`
> 
> I'll add the following info to the guide in the meantime:
> 
> Important
> 
> At the time of **2025-04-28** there appears to be a dependency error when installing Klipper. Does your Klipper installation fail with a `[ERROR] Error installing Python requirements: Installing Python requirements failed! [ERROR] Klipper installation failed!` please follow the next steps as a workaround and continue like normal:
> 
>     1. Start and do the Klipper installation like normal with KIAUH (new v6 beta 2 used in this example) until it exits with the above error. Quit KIAUH.
> 
>     2. Edit the file with: `nano ~/klipper/scripts/klippy-requirements.txt` and change `python-can==3.3.4` to **`python-can~=4.3`** (Save: CTRL-X, **Y**es)
> 
>     3. Start KIAUH again and (re)install Klipper (which is probably marked as 'incomplete' in the main screen by KIAUH). When asked to clone the repository and overwrite `/home/biqu/klipper` (already exists) say **No**, when asked to `Virtualenv already exists. Re-create?` say **Yes**.

Heads up, when **not** using KIAUH v6 the Klipper installation will error but KIAUH will act like Klipper installed successfully. This will lead to you not being able to install Klipper again (since it already is according to KIAUH) and when you remove Klipper with KIAUH you cannot edit the klippy-requirements.txt since it won't exist.

adamrodgers, 2025-04-28:
> Heads up, when not using KIAUH v6 the Klipper installation will error but KIAUH will act like Klipper installed successfully. This will lead to you not being able to install Klipper again (since it [...]

Well, shoot.  Looks like this is where I am right now.  Hoping someone has a quick fix.

Cogsworth03, 2025-04-28:
> Heads up, when **not** using KIAUH v6 the Klipper installation will error but KIAUH will act like Klipper installed successfully. This will lead to you not being able to install Klipper again (since [...]

Very very true! That's what was going wrong with my first two installations. Everything would act like it was going smoothly, but then clipper wouldn't be able to connect to moonraker, for obvious reasons. I'm glad there was at least a workaround. No I'm kind of scratching my head over the flashing of the MCU and tool head. About to take a break and restart that whole part of the process.

v-tron, 2025-04-28:
> Well, shoot. Looks like this is where I am right now. Hoping someone has a quick fix.

Just remove Klipper with KIAUH, then use v6 and the workaround that was posted earlier. Upon Klipper install with v6 you'll get the error, KIAUH recognizes it and aborts and you can edit klippy-requirements.txt and install again.

adamrodgers, 2025-04-28:
> Just remove Klipper with KIAUH, then use v6 and the workaround that was posted earlier. Upon Klipper install with v6 you'll get the error, KIAUH recognizes it and aborts and you can edit [...]

It worked and it looks like I'm up and running.  Thanks!