Rappetor/Sovol-SV08-Mainline #185: "Moonraker can't connect to Klipper!"

Cogsworth03, 2025-04-27:
So I'm trying to use the SD card method to switch to mainline but I've failed to do so twice now.
The ONLY issue I can see during the process is the warning/ error message I receive when installing Klipper itself. I've tried twice now.

I have flashed a couple of actual CB1's for my SV06+ in the past, but otherwise I'm new at it. Any help would be appreciated. 

[image]
[image]
[image]

Cogsworth03, 2025-04-27:
I just tried it for a third time on a new 8GB sd card (my other on is 64GB) and got an altogether different error... I have no clue what's going on or what I'm doing wrong.

[image]

chaz2000, 2025-04-27:
I currently have the same issue with "Moonraker can't connect to Klipper". I am doing it method 1, straight to the emmc. 
I had issues using KIAUH v6 as raised in #183 but solved it using v5.1.4. 
However, it looks like I'm getting the same issue as you. I'm going to try the steps suggested by Rappetor in #184.

chaz2000, 2025-04-27:
Sorry I ment KIAUH V5.1.4 instead of v6. On first run KIAUH asks if you want to use the new alph or the old one. I have had better luck using the older version. However it's not clear if it's a KIAUH issue or not because from my (limited) understanding it KIAUH only pulls in the files from elsewhere and has nothing to do with the final software. So in theory it shouldn't make a difference. 
I did try again today and the got a new issue with CB1 not being able to get updates. It looks like sometimes is happening in the background. Need to do some digging but I have spent too much time on it for now 😕

chaz2000, 2025-04-28:
I have just completed section 5, and the printer is now on and "happy"!!! (i statred over so there is no log of my connection issue) Thanks for adding the updates. I just got my Eddy Duo delivered today, so it's on to the next steps now. I will be taking an image of the Emmc at this point.