Rappetor/Sovol-SV08-Mainline #187: ls -la /dev/serial/by-id/ not working

chaz2000, 2025-04-29:
Hi, 
I just flashed the Katapult bootloader to the MCU and head bord. and moved on to step 8 however ls -la /dev/serial/by-id/ dose not apper to work. 

[image]

As you can see, I have installed pyserial. I have tried the normal reboot and double/triple checked the instructions, but I can not get it to work. Is this needed to check the mcu's are correct, or will this cause issues flashing Klipper?

thrystan, 2025-05-02:
I ran into that issue, and the reason was I overlooked the instruction to switch the mcu type from atmega to stm so the mcu wasn't booting.

chaz2000, 2025-05-04:
> I ran into that issue, and the reason was I overlooked the instruction to switch the mcu type from atmega to stm so the mcu wasn't booting.

This is exactly what I did. I'm now all booted and pushing plastic now!

closed 2025-05-04 by chaz2000