Rappetor/Sovol-SV08-Mainline #194: Automatic MCU update script will not work for Host MCU

ivanpetrov85, 2025-05-17:
When trying to flash Host MCU I had the script stopping at:
===============================
Connecting to Serial Device /dev/serial/by-id/usb-katapult_stm32f103xe_38FFD6053347533806742151-if00, baud 250000
Detected USB device running Katapul
================================

What fixed it to me was going to the web interface and stopping Crowsnest and Klipper.... from the "Power" ( top right) and the square "stop" next to each of them. 

Maybe add this to Step 8 of the manual notes.

[image]

ItsJust2You, 2025-05-18:
im  having the same issue but nothing seems to work ive disconnected everything i even changed the machines ip adres to make sure nothing is accidentally connecting to it but im still getting stuck on 

Host MCU firmware building complete. Press [Enter] to flash..

Flashing Klipper to HOST MCU.

Connecting to Serial Device /dev/serial/by-id/usb-katapult_stm32f103xe_38FFDA053348523720882451-if00, baud 250000
Detected USB device running Katapult

anyone have any other ideas?

thiagoshima, 2025-05-18:
Same, after a few hours stuck I was able to flash successfully using

`python3 flash_can.py -d /dev/serial/by-id/usb-katapult_stm32f103xe_50FF67068367515532431367-if00 -b 250000
`
 the script is in the katapult folder