Rappetor/Sovol-SV08-Mainline #195: automatic MCU update flash not working

ItsJust2You, 2025-05-18:
im having trouble getting klipper flashed on the board, the toolhead flash just fine in 1 try. I have closed all instances of mainsail and shutdown klipper and crowsnest just to be sure, i tried to disconnect the screen but still nothing. Anyone have any ideas?

Host MCU firmware building complete. Press [Enter] to flash..

Flashing Klipper to HOST MCU.

Connecting to Serial Device /dev/serial/by-id/usb-katapult_stm32f103xe_38FFDA053348523720882451-if00, baud 250000
Detected USB device running Katapult

anyone have any other ideas?

thiagoshima, 2025-05-18:
Same, after a few hours stuck I was able to flash successfully using

`python3 flash_can.py -d /dev/serial/by-id/usb-katapult_stm32f103xe_50FF67068367515532431367-if00 -b 250000
`
 the script is in the katapult folder

Missisippi882, 2025-08-11:
python3 flash_can.py -d /dev/serial/by-id/usb-katapult_stm32f103xe_50FF67068367515532431367-if00 -b 250000 
the script is in the katapult folder

@thiagoshima in which folder do you mean? I reading the problems. Sometimes not understanding is the description 7.2 wrong and we need 2x different katapult.bin?