Rappetor/Sovol-SV08-Mainline #197: Eddy-NG

maj0r1337, 2025-05-24:
Hi Richard,

do you plan to include eddy-ng to this repo? i cant find any useful info on how to set it up on our sv08 so i am kinda afraid to set it up by my self, at least with no info on how it works.

Best regards
Werner

khisanthax, 2025-08-03:
Thanks man, this saved me as well!

JeepRat2001, 2025-09-28:
Hi guys. I used ToppestOfDogs config files and plugged in my MCU id's. I'm getting this on startup: "Unknown pin chip name 'probe'". Only difference I can think of is that I installed Katapult on eddy first, then the firmware. The system sees it fine, but as:
Bus 003 Device 003: ID 1d50:6177 OpenMoko, Inc. rp2040

drwxr-xr-x 2 root root 100 Sep 28 08:18 .
drwxr-xr-x 4 root root  80 Sep 28 08:18 ..
lrwxrwxrwx 1 root root  13 Sep 28 08:18 usb-katapult_rp2040_504446677878121C-if00 -> ../../ttyACM2
lrwxrwxrwx 1 root root  13 Sep 28 08:18 usb-Klipper_stm32f103xe_32FFD5054747393913581557-if00 -> ../../ttyACM1
lrwxrwxrwx 1 root root  13 Sep 28 08:18 usb-Klipper_stm32f103xe_33FF6D064746373236270457-if00 -> ../../ttyACM0

Don't know is this could possibly be the problem, but Klipper seems to hang trying to connect to the eddy. Last night my log said it was timing out connecting. I'm thinking of reflashing eddy without katapult..??

[klippy-log-eddy-error.log](https://github.com/user-attachments/files/22585547/klippy-log-eddy-error.log)

JeepRat2001, 2025-09-29:
OK, I rebuilt and reflashed the eddy (still using Katapult) and now things look like I would expect:

lrwxrwxrwx 1 root root  13 Sep 28 22:17 usb-Klipper_rp2040_504446677878121C-if00 -> ../../ttyACM2
lrwxrwxrwx 1 root root  13 Sep 28 22:17 usb-Klipper_stm32f103xe_32FFD5054747393913581557-if00 -> ../../ttyACM1
lrwxrwxrwx 1 root root  13 Sep 28 22:17 usb-Klipper_stm32f103xe_33FF6D064746373236270457-if00 -> ../../ttyACM0

But still have the same error. I'll keep looking....