Rappetor/Sovol-SV08-Mainline #205: Error 3

jzagaja, 2025-07-30:
Got error 3 during print, signal cable new, extruder board new. 

they want me to use so called ST-Link burner: https://wiki.sovol3d.com/en/How-to-burn-firmware-to-the-motherboard-adapter-board
Sovol 3D Printer Wiki

please help

LexiconCode, 2025-07-31:
Are you using stock firmware or this repository?

jzagaja, 2025-07-31:
stock

LexiconCode, 2025-07-31:
> stock 

That guide looks very detailed. I don't think I could give you any better instruction. This isn't the best space for stock support. They have a discord channel which might have more people with the know-how.

jzagaja, 2025-07-31:
Been redirected from official discord here.

jzagaja, 2025-08-02:
I've changed nozzle and heater/thermistor so far no error. Stock nozzles leaking! Even those with additional screw.

jzagaja, 2025-08-03:
No still error 3.