Rappetor/Sovol-SV08-Mainline #208: issue with  flashing for hoat mcu

MattChew94877, 2025-08-08:
I have tried the entire proccess over and over again and every time im stopped by the same issue. when i try to flash the host mcu, it always freezes at: 

Connecting to Serial Device /dev/serial/by-id/usb-katapult_stm32f103xe_36FFD805334D413220551351-if00, baud 250000
Detected USB device running Katapult

i have made sure i follow every step perfectly and ive also tried the toolhead and that works perfectly. ive tried the solutions other people have used to fix this but they also dont work. ive tried manually flashing with the st link but then the printer no longer sees the host mcu.

Missisippi882, 2025-08-11:
Hello,

I have the same issue. Toolhead no problems for update.

Compiling out/src/stm32/chipid.o
Compiling out/src/generic/usb_cdc.o
Compiling out/src/stm32/hard_pwm.o
Preprocessing out/src/generic/armcm_link.ld
Building out/compile_time_request.o
PA1'own initial pin 'PA1
make: *** [Makefile:84: out/compile_time_request.o] Error 255
mv: cannot stat '/home/biqu/klipper/out/klipper.bin': No such file or directory

Host MCU firmware building complete. Press [Enter] to flash..

Flashing Klipper to HOST MCU.

Connecting to Serial Device /dev/serial/by-id/usb-katapult_stm32f103xe_36FFDB05334D413225731751-if00, baud 250000
Detected USB device running Katapult

Missisippi882, 2025-08-11:
Hello MattChew94877,

I fixed it, when you flash mcu with the katapult.bin under 7.7 is the picture delete correct with marking verify programming and full chip erase set. When you downloading the katapult.bin it is important the size correct setting 0x1188!! that is red marking.

Than Flash. It coming the same error info but it is running.

When in your printer.cfg the usb-katapult.....change to usb-Klipper! My mistake was I writing klipper not large and than the mcu not detected in the dashboard. Now my Sovol SV08 hope is ready.

My next step is fixing crownsnest that was under kiauh not possible..

LovesTha, 2026-01-05:
@Vint232 We clearly have the script as the failure is while running the script.

LovesTha, 2026-01-05:
How did you copy it?

MattChew94877, 2026-01-05:
the same method it says to backup your original stock klipper just to that and then install that file into the toolhead like you would installing katapult

LovesTha, 2026-01-05:
Cool, except my stlink only worked long enough for the two installs of Katapult, so I'm waiting on another new one :(

LovesTha, 2026-01-05:
My issue must be a tiny bit different to yours, I can use the script to install to the tool head but not the main mcu.

I think the simpler thing is to use the script to build the mainboard MCU lipper image, but diverge from the instructions by not selecting the 8kb offset, then use the stlink to flash that directly to the mainboard MCU.

Or is it that once you use stlink to copy the katapult + klipper to the other MCU the script starts to work fine?

MattChew94877, 2026-01-05:
oh our issues are completely reversed then. you could try copying klipper from the toolhead and install it on your main mcu. it might work but i honestly have no clue

Vint232, 2026-01-06:
> [@Vint232](https://github.com/Vint232) Matt is suggesting using the same Klipper image on both, which this repo and the update script don't recommend.
> 
> But I think his method (flashing katapult + klipper with stlink) is a viable idea, but it makes more sense to just flash klipper with stlink. The 8kb offset is to load klipper with katapult, if I can't successfully update via katapult then I might as well try raw klipper without the 8k offset.

You can't do it becouse it is different configuration. at this you hawe 2 mcu or 2 toolheads. main  katapult.bin it is boot loader! then using skript you select configuration of cpu and pins  according to appoint

LovesTha, 2026-01-06:
Before flashing Klipper they have katapult IDs, nice for the video to confirm that. The instructions could do with being clearer about that difference.

But the video just confirms I did things right, there isn't meant to be a difference in procedure in backup up the old FW or flashing katapult.

LovesTha, 2026-01-06:
Also I feel the flashing of katapult did go according to plan, I had two `usb-katapult-etc` devices in `/dev/serial/by-id/`. So the firmware is evidently running. And I used the same command line to flash both (not using Cube) and one was able to be flashed fine, it really should work.

As it is the mainboard mcu that wont flash, it isn't even that there could be an issue with a cable being loose.

Vint232, 2026-01-06:
> Okay, I just tried for no good reason to try the flashing of the mainboard MCU with the USB cable to the hot end disconnected.
> 
> It worked.
> 
> Computers suck!

Did it finally work?
If everything is going well, it's a wrong way :)