Rappetor/Sovol-SV08-Mainline #211: Error during Step 4.1

JeepRat2001, 2025-08-15:
I get these errors during repeated attempts starting from the beginning (using eMMC):

biqu@BTT-CB1:~$ sudo apt update && sudo apt upgrade
Get:1 http://deb.debian.org/debian bullseye InRelease [75.1 kB]
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [44.0 kB]
Ign:3 http://deb.debian.org/debian bullseye-backports InRelease
Get:4 http://deb.debian.org/debian-security bullseye-security InRelease [27.2 kB]
Err:5 http://deb.debian.org/debian bullseye-backports Release
  404  Not Found [IP: 151.101.42.132 80]
.........

Reading package lists... Done
E: The repository 'http://deb.debian.org/debian bullseye-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Repository 'http://deb.debian.org/debian bullseye InRelease' changed its 'Version' value from '11.9' to '11.11'
N: Repository 'http://deb.debian.org/debian bullseye InRelease' changed its 'Suite' value from 'oldstable' to 'oldoldstable'
N: Repository 'http://deb.debian.org/debian bullseye-updates InRelease' changed its 'Suite' value from 'oldstable-updates' to 'oldoldstable-updates'
N: Repository 'http://deb.debian.org/debian-security bullseye-security InRelease' changed its 'Suite' value from 'oldstable-security' to 'oldoldstable-security'
biqu@BTT-CB1:~$

Are these issues show stoppers? I've continued on after these issues, but haven't been able to get a bootable printer yet.

?????? Help ?????

JeepRat2001, 2025-08-15:
Crowsnest Install:

Fails at the same bullseye-backports Release...

Running apt-get update first ...

Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye-updates InRelease
Ign:3 http://deb.debian.org/debian bullseye-backports InRelease
Hit:4 http://deb.debian.org/debian-security bullseye-security InRelease
Err:5 http://deb.debian.org/debian bullseye-backports Release
  404  Not Found [IP: 151.101.42.132 80]
Reading package lists...
E: The repository 'http://deb.debian.org/debian bullseye-backports Release' no longer has a Release file.
Running apt-get update first ... [FAILED]

closed 2025-08-15 by JeepRat2001

JoewagKld, 2025-08-22:
https://github.com/Rappetor/Sovol-SV08-Mainline/issues/207#issuecomment-3198469188