Rappetor/Sovol-SV08-Mainline #216: Unable to add webcam

johnnynoir, 2025-09-18:
I've completed the upgrade to mainline klipper and all seems to be working with the exception of the adding the webcam. I'm seeing this error: "Error while connecting to http://192.168.1.213/webcam/?action=snapshot".

JeepRat2001, 2025-09-18:
Google search A.I. has the answer, just did that today... add the camera again under devices.

Yahoo Mail: Search, Organize, Conquer

johnnynoir, 2025-09-18:
That is when I see the error message. And yes, I've tried rebooting after that with no change. I tried to trace the webcam cable to see if possibly it'd become disconnected during the upgrade to mainline. No dice there since it's nearly impossible to do so with some major disassembly. Any idea about where the webcam is connected?